patman15/aiobmsble #7: dbus-serialbattery (Victron Energy Venus OS) integration

mr-manuel, 2025-09-06:
Reference to: https://github.com/patman15/BMS_BLE-HA/issues/475

On a complete new system with Python 3.12 I needed to additionally install this packages:
- [`bleak-retry-connector`](https://pypi.org/project/bleak-retry-connector)
- [`bluetooth-adapters`](https://pypi.org/project/bluetooth-adapters)
- [`aiooui`](https://pypi.org/project/aiooui)

Now I'm able to read the values of my JKBMS and it seems to instantly work better then with my current implementation.

Some useful data is missing, for example if the charging or discharging is allowed or not. Is it ok for you, if I will open a PR somewhen in the future to add this data for the BMS where I know how to fetch this data?

For all other BMS this data would remain `None`.

Some examples:
- Jikong BMS: https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/master/dbus-serialbattery/bms/jkbms_ble.py + https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/master/dbus-serialbattery/bms/jkbms_brn.py
- Kilovault: https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/master/dbus-serialbattery/bms/kilovault_ble.py
- LiTime: https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/master/dbus-serialbattery/bms/litime_ble.py
- LLT/JBD BMS: https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/master/dbus-serialbattery/bms/lltjbd.py + https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/master/dbus-serialbattery/bms/lltjbd_ble.py

How do you continuisly poll the data with a reconnection in case of an error? What is the polling intervall you are using or are you maybe using a callback from the BMS?

mr-manuel, 2025-09-06:
My goal would be:
1. Scan for BLE BMS nearby and create a list with names and MAC addresses
2. Connect to all BLE BMS and poll/receive data from them at max every 5 seconds (here I have to check if parallel or serial is better for CPU ressources)

mr-manuel, 2025-09-07:
Thanks for all that answers, they are helping me a lot!

> I can open a branch and you can then raise your PR against this branch

Sounds great to me. Unfortunately I can only contribute to the BMS which I already have in my driver. I have to check, if we can use maybe additional informations from the serial drivers I have.

> For best performance, I keep the connection open...

I downloaded the current main branch where you merged these changes. Now I was able to make it work and poll data without reconnect. With the Jikong BMS I have a beep every time I poll the data, which means a beep every 15 seconds in my case.

When using the JKBMS app there are two consecutive beeps when you first connect and then one beep if you switch between status, settings and control page. If you stay on the status page you have no more beeps and the data is continuously refreshing.

With dbus-serialbattery the BMS beeps only when connecting. Is there a way to improve that or am I doing something wrong? I used your code of the [README.md](https://github.com/patman15/aiobmsble/blob/0f57471cd66d50223c17a47ec1fe0fe0aa3951e2/README.md?plain=1#L58-L59) and added a loop with sleep:

```python
...
            while True:
                data: BMSsample = await bms.async_update()
                logger.info("BMS data: %s", repr(data).replace(", ", ",\n\t"))
                sleep(15)
...
```

For dbus-serialbattery this part of the [code](https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/master/dbus-serialbattery/bms/jkbms_brn.py) was written by someone other then me, I have no experience with BLE connections and data polling. That is also why I'm stuck on making that work smoothly with dbus-serialbattery and replacing the existing code with your library would be a good solution.

> Polling is by default 30s in Home Assistant. I think you could go down to ~10s

I think 15 seconds should be enough in that case, thanks!

> Typical Bluetooth adapters limit the number of parallel connections to a maximum of 3-5 if they are good ones.

If I check the page [known-working-high-performance-adapters](https://www.home-assistant.io/integrations/bluetooth/#known-working-high-performance-adapters) and search for some of this adapters to buy, none are available. Are they maybe superseeded with a newer chip or is the newer chip not working reliable?

mr-manuel, 2025-09-09:
Perfect, thanks!

> You could use the MAC address

Indeed, I will make that.

> No idea, but it also does not really matter as long as it works. 😄 So I would say what you experience is known behaviour, let me know if there are issues.

Unfortunatelly it matters, since users have their battery in the van or motorhome. You can't sleep with that noise every 15 seconds while sleeping 😄 I will check if I can fix that somehow or would you be able to check shortly if you spot the relevant difference between my and your code? Unfortunately your Discord link is not working, can you contact me via my Discord link?

> You can't (see temperature topic above) because that is really difficult to identify

In my case this is problematic. The MOSFET temperature sensor normally has a higher temperature then the cells. If for any reason a cell temperature rises to 40° C, then this could be very problematic. If the MOSFET temperature is 40° C then it's just chilling. With my driver I'm reducing charge/discharge current if temperature rises too high. Maybe we can add `temp_mos_values` where the MOSFET temperatures are added, if they are identified? That would allow me to know the MOSFET temperatures and then remove them from the `temp_values` list on my side. Would that work for you or do you have a better idea, without breaking the current functionallity of your library?

mr-manuel, 2025-09-10:
> Ok, but which code does the permanent beeping? Mine or yours, I somehow lost track ...

Yours (the code from the `aiobmsble` library)

> Can you tell me the SW version of your JKBMS?

Hardware version: V15A, Software version: V15.29

mr-manuel, 2025-09-16:
> Does this mean all BMS supported by [BMS_BLE-HA](https://github.com/patman15/BMS_BLE-HA) will be supported by GX-devices?

Yes

> Will VictronOS recognise my Creabest Battery on CerboGX?

If the library supports it, then yes.

snipah, 2025-09-16:
Yes, BLE-BMS supports it. Would be great to test. Let me know if I can support

mr-manuel, 2025-11-06:
Would be nice, if you can integrate the BMS that are not available in your library.

- Kilovault: https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/master/dbus-serialbattery/bms/kilovault_ble.py
- LiTime should be LiPower, right?
  https://github.com/mr-manuel/venus-os_dbus-serialbattery/blob/master/dbus-serialbattery/bms/litime_ble.py

patman15, 2025-11-06:
LiTime is Redodo and should work already now. I'll do an issue for the Kilovault and see how difficult it is to get tests for it.

mr-manuel, 2025-12-17:
As long as this issue is still open, it‘s not available: https://github.com/mr-manuel/venus-os_dbus-serialbattery/issues/313#issuecomment-3307647054