patman15/aiobmsble #30: Ective BMS not detected

techcraftde, 2025-09-19:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this library is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous enhancement requests](https://github.com/patman15/aiobmsble/issues?q=is%3Aissue%20label%3Aenhancement%20OR%20label%3A%22new%20device%22).
- [x] I have tried the [latest version of aiobmsble](https://pypi.org/project/aiobmsble/).

### Please describe the enhancement you would like to have.

Hi, 
Please integrate the Ective BMS with the uid 
0000af30-0000-1000-8000-00805f9b34fb
Currently the battery is not discovered.
Thank you!

patman15, 2025-09-20:
Hi! can you please 
* Provide the advertisement data of the device. (Run `aiobmsble` from the command line or in Home Assistant, go to the [advertisement monitor](https://my.home-assistant.io/redirect/bluetooth_advertisement_monitor/), click the device in question, and provide the information via **`copy to clipboard`**). 
* The information from the [service explorer script](https://github.com/hbldh/bleak/blob/develop/examples/service_explorer.py) 
Thanks!

techcraftde, 2025-09-20:
Hi, 

Advertisement Monitor:

{"name":"23-01-05-034","address":"10:ED:FA:7D:04:48","rssi":-52,"manufacturer_data":{},"service_data":{},"service_uuids":["0000af30-0000-1000-8000-00805f9b34fb"],"source":"B8:27:EB:8B:0B:98","connectable":true,"time":1758351230.3644323,"tx_power":null,"raw":"020106030330af"}