patman15/aiobmsble #36: Expose connect() and is_connected

fl4p, 2025-09-29:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] I have understood that this library is only about battery management systems with Bluetooth Low Energy support.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate of any [previous enhancement requests](https://github.com/patman15/aiobmsble/issues?q=is%3Aissue%20label%3Aenhancement%20OR%20label%3A%22new%20device%22).
- [x] I have tried the [latest version of aiobmsble](https://pypi.org/project/aiobmsble/).

### Please describe the enhancement you would like to have.

Sometimes we cannot use context managers and like to probe a BMS connection, so we need a connect() function.
Also a `is_connected` property (which would resolve to `bms._client.is_connected`) would be useful

patman15, 2025-09-30:
Hi! Can you explain the use-case a bit more? Not sure why the usage of the library would require those functions.
You can see a usage of the library without context manager in [\_\_main\_\_.py](https://github.com/patman15/aiobmsble/blob/main/aiobmsble/__main__.py). The library takes care of all connection topics, you just need to call `async_update()`. By default connections are reused and re-established if needed, so no need to query via `is_connected`. You just need to call disconnect before dropping the object.
This also means the object is always connected if you did not set `keep_alive=False` on `init()`.

closed 2025-11-10 by github-actions[bot]

fl4p, 2025-11-10:
`is_connected`: This would give the user the freedom to implement custom re-connect logic and the ability to monitor the link status.
In my application I keep a log of the connection status.

I see that instead of `connect()` an `async_update()` would work.